PepperlFuchs/pf_lidar_ros_driver #1: Not getting any data from any of the scan topics

milano-technical-intern, 2019-09-27:
Hi,

I'm not getting any data from any of the scan topics, i.e, by executing rostopic echo /r2300_node/scan_0 or by trying to visualize them using RViz. I'm not getting any error either when executing the launch file so, I'm not sure what is happening. I'm using the default values: 

SUMMARY
========

PARAMETERS
 * /r2300_node/frame_id: /scan
 * /r2300_node/major_version: 0
 * /r2300_node/samples_per_scan: 500
 * /r2300_node/scan_frequency: 100
 * /r2300_node/scanner_ip: 10.0.10.76
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /
    r2300_node (pf_driver/r2300_node)

and these are the topics I'm getting:

/r2300_node/scan_0
/r2300_node/scan_1
/r2300_node/scan_2
/r2300_node/scan_3
/rosout
/rosout_agg

By the way, I'm able to ping the R2300 with no problems. [Here](https://photos.google.com/share/AF1QipNrHek4WQ5wtgsVd29fRLrXIet5Q1X9ikqR6NzE7e9c6dY4X9lkHOczMQ5KUTCmRw?key=UlpoZHBjaGpQc1VicDgtTGE2dXZWQXBwMGZ0OE13) is a picture of the status of the LEDs.

Best,

ptruka, 2019-09-30:
Hello,

a quick question to begin: Which hardware version are you using?

Best Regards

milano-technical-intern, 2019-09-30:
This is the information related to the R2300 I have:

HW: 0.608
FW: 0.70

Best,

ipa-hsd, 2019-10-01:
@milano-technical-intern could you please pull the latest master and try again?

milano-technical-intern, 2019-10-03:
@ipa-hsd I did, and I'm still having the same issue: no data from any of the scan topics. Have you updated the master @ipa-hsd? In the repository, it reads "Latest commit adff7b9 13 days ago"

Best,

ipa-hsd, 2019-10-07:
@milano-technical-intern yes I have updated the master branch. Have you made sure the device is in working condition? Probably just pinging it may not be enough. @ptruka is there any Windows software where he can test the device?

milano-technical-intern, 2019-10-08:
@ipa-hsd I just tested it out in the Windows 10 program. It works. I can see all four layers. The problem persists on the ROS driver. I'm not getting any data. Do you know what indicates each LED in the R2300? It seems that in the ROS program is not establishing connection with the R2300. As you can see in [this picture](https://photos.google.com/share/AF1QipNrHek4WQ5wtgsVd29fRLrXIet5Q1X9ikqR6NzE7e9c6dY4X9lkHOczMQ5KUTCmRw?key=UlpoZHBjaGpQc1VicDgtTGE2dXZWQXBwMGZ0OE13), the orange LED keeps blinking when using the ROS driver. On the other hand, when using the Windows program is not blinking at all.

Best,

milano-technical-intern, 2019-10-10:
Nevermind. I just got the manual where it describes the indicators.

The PC where I'm testing the R2300 doesn't have WiFi. Therefore, when I connect to the R2300 is by Ethernet only. I've tried with a network switch and directly connecting the R2300 to the computer but none of them worked.

ipa-hsd, 2019-10-10:
OK seems like it's a network issue. I'll close the issue for now. Feel free to open it again.

milano-technical-intern, 2019-10-10:
I'm getting the same result with or without the network switch. I even tried with another computer and it's the same.

Best,

milano-technical-intern, 2019-10-10:
@ipa-hsd by the way, I cannot reopen the issue because I'm not a collaborator since you closed it.

https://stackoverflow.com/questions/21333654/how-to-re-open-an-issue-in-github

ipa-hsd, 2019-10-15:
I will try to implement SSDP within the driver, that might solve the issue.

milano-technical-intern, 2019-10-17:
@ipa-hsd  Have you tried to use the driver in the freshly installed Ubuntu 18.04 and ROS melodic? Maybe you can replicate the issue I'm having by doing that.

Best,

YangAndI, 2019-10-18:
I have the same problem.
Infinite loop
  data_notifier.wait_for(lock, std::chrono::seconds(1));
because the scans is empty
best

YangAndI, 2019-10-18:
And the led is only three,the first is green and the last is green,and the second to last is flashing yellow light.
Best,

ipa-hsd, 2019-10-18:
@YangAndI and @milano-technical-intern is the scanner's IP `10.0.10.76` and your host PC's IP `10.0.10.1`?

milano-technical-intern, 2019-10-18:
@ipa-hsd Actually, It worked by changing the host PC's IP to 10.0.10.1

Thanks!

Best,

YangAndI, 2019-10-21:
@ipa-hsd Thanks,it is worked by changing the IP to 10.0.10.1.
May i change the sensor ip or host ip?

milano-technical-intern, 2019-10-31:
@ipa-hsd Any particular reason why it doesn't work for any other address in the subnet but 10.0.10.1? If I send HTTP commands like this one:
`http://10.0.10.76/cmd/set_parameter?pilot_laser=on`
it works for any address in the subnet.

Best,

ipa-hsd, 2019-11-11:
Sorry for the delay. Can you test with the latest `master`?

milano-technical-intern, 2019-11-15:
> Sorry for the delay. Can you test with the latest `master`?

Still not working with other ip address other than 10.0.10.1

ipa-hsd, 2019-11-28:
Make sure your IP is has a subnet of 10.0.10.*    
I cannot reproduce this error from my side anymore.

bk-mtg, 2019-12-05:
@ipa-hsd I work with @milano-technical-intern, and I just finished testing that this appears to be working now for us.  We even verified that we are able to change the IP/subnet on the device (as expected) and the driver works on the new subnet (assuming appropriate host IP config obviously).

I assume this was addressed in c1760582afc0591cb0b4030d6b2a8d0b4a0da322?  Thanks for the help with this!

closed 2019-12-05 by milano-technical-intern

ipa-hsd, 2019-12-06:
Good to know it works!