PepperlFuchs/pf_lidar_ros_driver #2: Is UDP supposed to work?

maxlein, 2019-10-15:
Hi,

I am trying out the driver and I can't get it to work over UDP.  

First, there is a hardcoded udp address in `pfsdp_protocol.hpp`.  
Then at least parse_data() gets called but no scans are published...  

TCP is working.

ipa-hsd, 2019-10-15:
Hi, are you using R2000? UDP for R2000 is not tested yet. Is this a blocking issue for you?

maxlein, 2019-10-15:
Well, we are comparing the sick microscan against the r2000.  
And I see some time delay as you can see on the gif:  (the slow downs are from this gif recorder...)

[image]  

btw: it may occurred to you that r2000 has a quite high noise in comparison to the microscan.  
Do you have any suggestions for configuring the r2000 for localization tasks?

ipa-hsd, 2019-10-15:
> And I see some time delay as you can see on the gif: (the slow downs are from this gif recorder...)  

To clarify, the delay is not observed because of the driver, right?

> it may occurred to you that r2000 has a quite high noise in comparison to the microscan.

I have not tested SICK scanner you mentioned, so I cannot compare. Can you check if you get the same noise if you use the Windows software for Pepperl+Fuchs?

maxlein, 2019-10-15:
No, the delay is somewhere in the line from scanner to rviz.  
I was just referring to the gif not displaying smoothly.  

Something is buffering up, because after a restart of the driver the delay is gone but increases again as time is passing.

Yes, I have the same noise on Windows.

ipa-hsd, 2019-10-15:
> No, the delay is somewhere in the line from scanner to rviz.

This is most probably an issue with incorrect calculation of `time_increment`. Can you test https://github.com/PepperlFuchs/ROS_driver/commit/2f94e06e9a47030f625e6ea7f43e3b2e9a957d3b and let me know if this solves your issue?

ipa-hsd, 2019-10-15:
It is most probably an issue with incorrect calculation of `time_increment`. Can you test it with https://github.com/PepperlFuchs/ROS_driver/commit/0ee0024829ff4d0f6d27864ec84e2badb4b6aeec?

maxlein, 2019-10-16:
Yes, that fixed it! Thanks for fast support.

closed 2019-10-16 by ipa-hsd