PepperlFuchs/pf_lidar_ros_driver #3: Display custom text

maxlein, 2019-10-15:
Could not find a documentation which describes the interface, so I would like to know if it's possible to pass custom text which is displayed on the scanners display?

bk-mtg, 2020-07-27:
It *is* possible to pass this text (at least for R2000, not R2300 AFAIK).

If you're interested in doing it yourself via the [HTTP API](https://files.pepperl-fuchs.com/webcat/navi/productInfo/doct/doct3469f.pdf?v=20200506171824) (linked on the R2000 product pages as "R2000 Ethernet communication protocol"), you can start at `http://YOUR_R2000_IP_HERE/cmd/get_protocol_info`, which contains a list of other commands.  The commands of most interest for this application are `get_parameter` and `set_parameter`, for parameters `hmi_display_mode` and `hmi_{static,application}_text_{1,2}`.  For example, you might do `http://SCANNER_IP/cmd/set_parameter?hmi_display_mode=application_text&hmi_application_text_1=test&hmi_application_text_2=banner`.  The full protocol doc has lots more details, some of which I've found to be generally useful for operating the sensor even though they're not documented in the more general purpose manuals.

If you're interested in doing this via the ROS dynamic configuration API, first make sure you have the fixes I just pushed from #27 (or equivalent), then after starting the R2000 node, you should be able to `rosrun rqt_reconfigure rqt_reconfigure` and it will give you a list of parameters that you can edit.

ipa-hsd, 2021-01-13:
Feel free to reopen the issue if this is not solved for you.

closed 2021-01-13 by ipa-hsd