PepperlFuchs/pf_lidar_ros_driver #10: Error: Unknown parameter, 'measure_start_angle

maxlein, 2019-12-10:
When I start r2000 driver with latest code, I get this error on startup:  
Is my firmware not up to date or something?  

```
[r2000_node-2] protocol error: 0 success
[r2000_node-2] protocol error: 0 success
[r2000_node-2] [10/12/19 11:27:41:144:144954] [default] [info] UDP Connection: connecting to: 192.168.123.60:60849
[r2000_node-2] protocol error: 0 success
[r2000_node-2] protocol error: 110 Unknown parameter, 'measure_start_angle'.
[r2000_node-2] terminate called after throwing an instance of 'std::invalid_argument'
[r2000_node-2]   what():  stoi
```

maxlein, 2019-12-10:
Available parameters:  
```

{                                                                                                                                                                                                                            
"parameters":[                                                                                                                                                                                                               
"vendor",                                                                                                                                                                                                                    
"product",                                                                                                                                                                                                                   
"part",                                                                                                                                                                                                                      
"serial",                                                                                                                                                                                                                    
"revision_fw",                                                                                                                                                                                                               
"revision_hw",                                                                                                                                                                                                               
"max_connections",                                                                                                                                                                                                           
"feature_flags",                                                                                                                                                                                                             
"radial_range_max",                                                                                                                                                                                                          
"radial_range_min",
"radial_resolution",
"angular_fov",
"angular_resolution",
"ip_mode",
"ip_address",
"subnet_mask",
"gateway",
"scan_frequency",
"scan_direction",
"samples_per_scan",
"scan_frequency_measured",
"status_flags",
"load_indication",
"up_time",
"power_cycles",
"operation_time",
"operation_time_scaled",
"temperature_current",
"temperature_min",
"temperature_max",
"contamination",
"user_tag",
"device_family",
"mac_address",
"hmi_display_mode",
"hmi_language",
"hmi_button_lock",
"hmi_parameter_lock",
"ip_mode_current",
"ip_address_current",
"subnet_mask_current",
"gateway_current",
"system_time_raw",
"user_notes",
"locator_indication",
"scan_frequency_min",
"scan_frequency_max",
"sampling_rate_min",
"sampling_rate_max",
"hmi_static_text_1",
"hmi_static_text_2",
"hmi_application_text_1",
"hmi_application_text_2",
"hmi_static_logo",
"hmi_application_bitmap",
"operating_mode",
"emitter_type",
"filter_type",
"filter_width",
"filter_error_handling",
"filter_remission_threshold",
"filter_maximum_margin"
],
"error_code":0,
"error_text":"success"
}

```

ipa-hsd, 2019-12-10:
This is not a firmware issue. I will get back to you regarding this.

Patrick-Lascombe, 2020-02-26:
measure_start_angle is a parameter for the r2300 LIDAR. Comment the line `angle_min_max = protocol_interface->get_angle_min_max(handle_info.handle);` in pf_interface.hpp and it should work.

bk-mtg, 2020-07-27:
I was successfully able to capture with an R2000 this week and did not experience this issue, so I think this has been resolved?

ipa-hsd, 2020-07-27:
Yes, it was fixed in https://github.com/PepperlFuchs/ROS_driver/pull/14

closed 2020-07-27 by ipa-hsd