PepperlFuchs/pf_lidar_ros_driver #16: Too many scans in receiver queue: Dropping scans!

Heavydpan, 2020-06-05:
Hi!
My customer start R2000 and start to get R2000 point data, but after 1 hour, the R2000 driver post err:Too many scans in receiver queue: Dropping scans!
They use TCP request to get R2000 data.

How to solve this issue?
Thannk you!

ipa-hsd, 2020-06-05:
@Heavydpan you can ignore the message. It just says that there are too many scans (100 in this case) in queue and not all of them will be processed. Probably I will remove the log message if it is annoying.

Heavydpan, 2020-06-05:
But after this message showed, they can't receive any data.

Is it other issue?

ipa-hsd, 2020-06-05:
OK then it is an issue. I don't think not receiving data is directly related to the log message.

Heavydpan, 2020-06-05:
Hi,
But after this message showed, they can't recieve any data.
Is it other issue?

NikolasE, 2020-06-22:
"not all of them will be processed" Does that mean that not all measured data will be available in subscribing nodes?

ipa-hsd, 2020-06-22:
Yes the device sends data at a faster rate than the ROS driver can process it. So the driver maintains a queue internally. Once the queue becomes full, oldest scan data is dropped.

NikolasE, 2020-06-22:
How many scans are getting dropped? And how big can the delay between capturing the data on the device and publishing it as laser-msg become? I think that removing the warning is not the right way to deal with that problem as it hides a potentially serious issue.

ipa-hsd, 2020-08-06:
@Heavydpan there have been changes in the past couple of months. Could you please pull the `master` branch and try again?

ipa-hsd, 2021-01-13:
Feel free to reopen the issue if this is not solved for you.

closed 2021-01-13 by ipa-hsd