PepperlFuchs/pf_lidar_ros_driver #17: When call the httpGet(), the program is jam.

Heavydpan, 2020-06-05:
Sometimes the network is breakdown, and the customer track the driver, when run in HttpCommandInterface::httpGet(), and boost::asio::read_until() and boost::asio::read() will jam, and can't exit with CRTL+C.
We also can do like this: connect() to R2000, and sleep 5s, in this 5s run "ifdown" in terminal, have the same issue.
My suggestion is when the network have this problem , the R2000 driver throw the exception and return, maybe you can modify the driver?

Thank you!

ipa-hsd, 2020-08-06:
There have quite a few changes in the past couple of months. Could you please pull in the `master` branch and try again?

ipa-hsd, 2020-08-25:
I will close this for now. Feel free to open if you still face the issue.

closed 2020-08-25 by ipa-hsd