PepperlFuchs/pf_lidar_ros_driver #22: NaN range not filled in for packet type C

bk-mtg, 2020-07-18:
I haven't actually observed this in an operational setting, but I think this line that publishes NaN instead of the distance measurement only works for packet types A/B but not for packet type C.  Not sure if it matters in practice since in any case it will be above the `range_max` parameter and should be ignored, but it seems generally undesirable to treat the various packet types differently.
https://github.com/PepperlFuchs/ROS_driver/blob/477d72066f8cfc6ecb960fc388c2a8f0e9f6a3ec/pf_driver/src/ros/scan_publisher.cpp#L81-L82

ipa-hsd, 2020-08-06:
Is it address in https://github.com/PepperlFuchs/ROS_driver/pull/26?

bkurtz, 2020-08-17:
Not at present.

bk-mtg, 2020-08-17:
Not at present, no.