PepperlFuchs/pf_lidar_ros_driver #28: Driver status / release plans

reinzor, 2020-07-29:
Hi, 

What is the current status of the driver? Is a proper release planned anytime soon for `Melodic` / `Noetic`? According to https://github.com/PepperlFuchs/ROS_driver/issues/12 it was planned to be finished two months ago. 

Is there a way we can help?

ipa-hsd, 2020-07-29:
Thanks for your interest. There are still some minor bugs being fixed but the state of the driver has improved considerably since the issue you mentioned. We are currently targeting `Melodic`, but porting to` Noetic` shouldn't be much of an effort. Once it is ready for `Melodic`, I can create a new issue with TODOs to move to `Noetic` where you could chip in.

closed 2020-08-06 by ipa-hsd

reinzor, 2020-08-19:
Sorry for re-opening this issue but could you give more clarity about the release plan?

reinzor, 2020-08-25:
ping @ipa-hsd

ipa-hsd, 2020-08-25:
Once https://github.com/PepperlFuchs/ROS_driver/pull/26 is merged, the current development will come to an end for now. Then it would be ready for release.

reinzor, 2020-08-25:
Thanks for your reply :)

reinzor, 2020-09-07:
Hi @ipa-hsd , I see that the PR is merged. How far are we from an official release? Can we help in some way?

reinzor, 2020-09-30:
@ipa-hsd ping

reinzor, 2020-10-14:
@ipa-hsd, could you please provide some timeline or point in a direction where we can help? If not, we will start our own fork (which is ofcourse not the goal of any open source project).

ipa-hsd, 2020-10-15:
The PRs have been merged. The Pepperl-Fuchs team (of which I am not part of) is taking a decision about the release. It would be great if you could test the driver on Noetic until then and make a new issue with the errors you come across. Thanks!