PepperlFuchs/pf_lidar_ros_driver #32: Seg fault on ROS Noetic

Patrick-Lascombe, 2020-10-13:
Hello,
I know that this driver is not yet released on Noetic, but when I tried to use it I got a Seg fault (error code -11).
The driver is properly launched and I receive the LaserScab but after a few seconds the error (seen below) pop.
The error `conversion of data from string failed` comes from the parameter "skip_scans" missing since the LiDAR I use is not a version 1.03 or newer. But I don't think it is the cause of the problem since it has a default value of 0.
My setup is Ubuntu 20.04 with ROS Noetic.

[image]

ipa-hsd, 2020-10-13:
Thanks for reporting. Yes this seems to be because of the version mismatch of the device and not because of Noetic. Would it be possible for you to test the driver against a newer device version? I will try to print out a cleaner error message in the mean-time.

Patrick-Lascombe, 2020-10-13:
I'll see if we have a newer version in our offices and check it

Patrick-Lascombe, 2020-10-13:
Tested on a newer device, and the same problem occurred. I'll try to check on another computer during the day to exclude my hardware configuration
[image]

ipa-hsd, 2020-10-13:
Could you remove `skip_scans` from [this](https://github.com/PepperlFuchs/ROS_driver/blob/330a0425983c6b82f1b5c91e1f4d32b85b1d3075/pf_driver/include/pf_driver/pf/pfsdp_protocol.hpp#L385) line and comment out [this](https://github.com/PepperlFuchs/ROS_driver/blob/330a0425983c6b82f1b5c91e1f4d32b85b1d3075/pf_driver/include/pf_driver/pf/pfsdp_protocol.hpp#L390) line and try again?

Patrick-Lascombe, 2020-10-13:
Same result, I do not think it comes from the different version. I now conduct my test on a R2000 which has the skip_scans function and the same problem occurs

ipa-hsd, 2020-12-03:
I was able to build and run the code successfully in `Noetic`. Are you still having the issue? Could you please compile the code in `debug` mode 
```
catkin_make -DCMAKE_BUILD_TYPE=Debug
```
and run it with [gdb](http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB) (add the following to the `<node>` tag in your `launch` file)
```
launch-prefix="gdb -ex run --args" 
```

ipa-hsd, 2021-01-13:
Feel free to reopen the issue if this is not solved for you.

closed 2021-01-13 by ipa-hsd