PepperlFuchs/pf_lidar_ros_driver #34: ROS release

reinzor, 2020-11-11:
I've seen you created a release zip in Github. What about releasing the package using bloom so that it can be added to the `rosdistro`?

reinzor, 2020-11-11:
https://github.com/PepperlFuchs/ROS_driver/releases/tag/v1.0

ipa-hsd, 2020-11-11:
Yes they are in the process of doing it. It should be done in the coming weeks.

reinzor, 2021-01-04:
Another ping, we are waiting for an official ROS release :)

reinzor, 2021-01-07:
@PepperlFuchs could you please elaborate on this?

reinzor, 2021-01-19:
@ipa-hsd , @PepperlFuchs could you please elaborate on the plan with this driver? If not, we have no other choice than forking this driver and maintain a fork (which is obviously not desired for an open-source project).

ipa-hsd, 2021-01-19:
@reinzor sorry for the delay, but there are some pending PRs which we are trying to merge ASAP. Since they are fixes to existing (minor) bugs, it would not make much sense to release the packages without these fixes (the driver "works" without these fixes though).

ipa-hsd, 2021-02-09:
Finally it is done:  
https://github.com/ros/rosdistro/pull/28271  
https://github.com/PepperlFuchs/ROS_driver/releases/tag/1.1.0   

It takes few days till it is indexed and then you should be able to install the package. Thanks for your patience!

closed 2021-02-09 by reinzor