PepperlFuchs/pf_lidar_ros_driver #35: Bug: dynamic reconfigure

jadhm, 2020-11-30:
Hi, 

I think there is a bug in the dynamic reconfigure. Once you update a parameter all the other parameters get updated to the default value. More specifically, in this line you re-update the the values from the output of the scanner and here there is something going wrong that reset the values of the parameters. In addition, I noticed that changing the config values for example the _start_angle_ has no impact because the values are always overwritten due to this line of code. By deleting this line the dynamic reconfigure works as it should be but I am wondering why do you re-write the values from the output of the scanner ? 

https://github.com/PepperlFuchs/ROS_driver/blob/8cf3b8dc8121aa0c0dc30e34da8685326ee52b7f/pf_driver/src/pf/pf_interface.cpp#L254

wsaihopfsg, 2021-01-05:
I think what is missing is a `set_scanoutput_config` that actually communicates the config changes to the R2000/R2300.

ipa-hsd, 2021-07-30:
Solved in https://github.com/PepperlFuchs/pf_lidar_ros_driver/pull/49

closed 2021-07-30 by ipa-hsd