PepperlFuchs/pf_lidar_ros_driver #37: feat: add time_offset parameter

jadhm, 2020-11-30:
Hi, 

I sent a pull request  #36 to add a new parameter time_offset in case there is latency. 

Regards,
Jad

ipa-hsd, 2020-12-01:
https://github.com/PepperlFuchs/ROS_driver/pull/36#issuecomment-736545604

jadhm, 2021-03-18:
@ipa-hsd  could you please take a look at the pull request ?

jadhm, 2021-03-22:
ping @ipa-hsd

ipa-hsd, 2021-03-22:
Though time offset is an issue, but the solution is not acceptable. It does not make sense to add a user-defined time offset.

jadhm, 2021-03-25:
@ipa-hsd  I see your point but if you looked at other drivers of other sensors you would see that they also have this parameter for solving latency. This is an example of the sick driver 
https://github.com/SICKAG/sick_safetyscanners/blob/cfa456e78af42b82c1031354864837a798add4a7/config/SickSafetyscannersConfiguration.cfg#L20

ipa-hsd, 2021-03-25:
Yes I have seen that and the realized that was the basis of this PR. But does not necessarily mean it is a good solution.

ipa-hsd, 2021-03-25:
I might be wrong. According to the [REP](https://www.ros.org/reps/rep-0138.html), it does mention `calibrate_time` and `time_offset`.

jadhm, 2021-03-26:
@ipa-hsd  dealing with latency is a bit tricky but a parameter for fine tuning should be possible. Would you be open for re opening my PR ?

reinzor, 2021-03-26:
@ipa-hsd , just to clarify: Nobody is alterint the `ros::Time`. The `ros::Time` is determined by the system clock if `/use_sim_time` parameter in the global namespace is `false`. If `true`, the `ros::Time` is determined by a `ClockPublisher` that publishes on the `/clock` topic. This additional `time_offset` parameter is only required for manually adjusting the timestamps of the incoming sensor readings. If `ros::Time::now()` is used, it means that the sensor reading was created at the current time. However, this is ofcourse not the case since the message was created x seconds ago due to a acquisition and communication delay. The `time_offset` in sensor drivers is often introduces to cope for these "unmeasurable" delays. Hope this clears things up.

reinzor, 2021-04-12:
@ipa-hsd any updates on this?