PepperlFuchs/pf_lidar_ros_driver #38: Reconnictivity issue

jadhm, 2020-12-01:
Hi, 

The the node reconnects once the connection is lost with the scanner but the node doesn't keep trying to reconnect. In other words, if the connection was lost for a particular time the node raises an errors and it doesn't recover at all which means that the re-connectivity is not robust. Would you please look at this problem because the node should always keep trying to reconnect. 

Regards,
Jad

ipa-hsd, 2020-12-01:
Yes I am aware of this and added an `enhancement` label to the issue.

jadhm, 2021-03-18:
@ipa-hsd  Hi ! Any updates about this issue ?

ptruka, 2021-03-29:
@jadhm Hi, this enhancement will be included in the next major version.

ipa-hsd, 2022-07-11:
https://github.com/PepperlFuchs/pf_lidar_ros_driver/pull/81 has been merged

closed 2022-07-11 by ipa-hsd