PepperlFuchs/pf_lidar_ros_driver #42: R2300 can't visualize point cloud in rviz

john-sokko, 2020-12-03:
Launching _r2300.launch_, I get the following error in rviz for the ``PointCloud2`` display: 

``Failed to transform from frame [/base_link] to frame [base_link]``

``base_link`` is selected as the Global Fixed frame.  

I checked the ``/cloud`` topic at the command line, and it is publishing at 10-11 Hz.  I can also visualize each of the 4 ``LaserScan`` topics.  

Any ideas how to fix?

john-sokko, 2020-12-03:
Here is a screenshot.
[image]

ipa-hsd, 2020-12-04:
Is this on `Ubuntu 20.04` and `ROS Noetic`? I am having the same problem on `Noetic`.

john-sokko, 2020-12-04:
Yes.

ipa-hsd, 2020-12-04:
looks like it is a known error: https://answers.ros.org/question/355595/rviz-tf-error-failed-to-transform-from-frame-map-to-frame-map/

wsaihopfsg, 2020-12-09:
I can successfully invoke `roslaunch pf_driver r2300.launch`, but cannot see anything in rviz.

[image]

```
... logging to /home/pepperlfuchsctss/.ros/log/c02be5fa-3a02-11eb-9f3e-00155d97d3b7/roslaunch-1806PTN523-2997.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://1806PTN523:37105/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    r2300_node (pf_driver/ros_main)
    ring1 (tf/static_transform_publisher)
    ring2 (tf/static_transform_publisher)
    ring3 (tf/static_transform_publisher)
    ring4 (tf/static_transform_publisher)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[r2300_node-1]: started with pid [3012]
process[ring1-2]: started with pid [3013]
process[ring2-3]: started with pid [3014]
process[ring3-4]: started with pid [3015]
process[ring4-5]: started with pid [3025]
[ INFO] [1607515719.099464700]: Device found: OMD10M-R2300-B23-V1V1D-4S
process[rviz-6]: started with pid [3037]
[ INFO] [1607515719.173835000]: Device state changed to Initialized
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-pepperlfuchsctss'
[ INFO] [1607515719.208733100]: Device state changed to Running
```
I checked all the cloud related topics `/cloud_1 /cloud_2 /cloud_3 /cloud_4`, all are not publishing.

The only topic that seems to be actively publishing is `/tf`.

The unit is not defective since I can visualize using another P+F software.

I am running WSL with `Ubuntu 18.04` and `ROS Noetic`. 

Thank you.

ipa-hsd, 2020-12-09:
@wsaihopfsg check the link posted in this thread. It is apparently a known issue in Noetic.

wsaihopfsg, 2020-12-09:
Sorry I made a typo, I am using melodic

ipa-hsd, 2020-12-09:
It would be difficult to replicate this error since there has not been an issue with `Melodic` before and there seems to be no error output in your case. Can you check the output of the pointcloud and laserscan on terminal with
```
rostopic echo /cloud
rostopic echo /cloud_1
```

wsaihopfsg, 2020-12-09:
Both `/cloud` & `/cloud_1` are not publishing. In fact all topics are not publishing except for `/tf`, `/r2300_node/parameter_descriptions` & `/r2300_node/parameter_updates`

Not sure if it's related but I realized that with the web browser `http://192.168.0.78/cmd/get_parameter`, my settings are correct
```
"ip_mode_current":"static",
"ip_address_current":"192.168.0.78",
"subnet_mask_current":"255.255.255.0",
"gateway_current":"192.168.0.10",
```
But when I  `rostopic echo /r2300_node/parameter_updates`, I get instead
```
strs:
  -
    name: "ip_mode"
    value: "autoip"
  -
    name: "ip_address"
    value: "10.0.10.76"
  -
    name: "subnet_mask"
    value: "255.0.0.0"
  -
    name: "gateway"
    value: "0.0.0.0"
  -
```

closed 2021-01-13 by ipa-hsd