PepperlFuchs/pf_lidar_ros_driver #43: cURLpp not found on Debian 9.0

wsaihopfsg, 2020-12-08:
Hi,

I tried installing the driver on Debian 9.0. 

During the rosdep install step, `Unable to locate package libcurlpp-dev` is reported.

I proceeded with an alterative package `apt-get install libcurl4-openssl-dev`. 

In the CMakeFiles.txt I added `pkg_check_modules(CURL libcurl REQUIRED)`.

However during catkin_make compilation all `#include <curlpp/cURLpp.hpp>` will get an "No such file or directory" error.

I guess the CMakeFiles.txt is still broken but I have no idea how to correct it.

Anyway here is the output for `pkg-config --print-variables libcurl`
```
exec_prefix
includedir
libdir
pcfiledir
prefix
supported_features
supported_protocols
```

Thank you.

ipa-hsd, 2020-12-08:
That is strange because the key for `debian` is `libcurlpp-dev` as seen [here](https://github.com/ros/rosdistro/blob/3b3b962005fca9a6032fb99ec9b07ee3631e0ad5/rosdep/base.yaml#L618). I don't think `libcurl4-openssl-dev` is the same package so you are bound to get `No such file or directory` error. Are you sure you have done
```
sudo apt-get update
rosdep update
```

wsaihopfsg, 2020-12-08:
Yes I have performed both commands.

While `libcurlpp-dev` was not found, this was the message for `sudo apt-get install libcurl-dev`, and subsequently I chose `libcurl4-openssl-dev`
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package libcurl-dev is a virtual package provided by:
  libcurl4-openssl-dev 7.52.1-5+deb9u12
  libcurl4-nss-dev 7.52.1-5+deb9u12
  libcurl4-gnutls-dev 7.52.1-5+deb9u12
You should explicitly select one to install.
```

ipa-hsd, 2020-12-08:
These are some of the commands I used:
```
$ dpkg -l | grep curl

ii  curl                                            7.68.0-1ubuntu2.2                           amd64        command line tool for transferring data with URL syntax
ii  libcurl3-gnutls:amd64                           7.68.0-1ubuntu2.2                           amd64        easy-to-use client-side URL transfer library (GnuTLS flavour)
ii  libcurl4:amd64                                  7.68.0-1ubuntu2.2                           amd64        easy-to-use client-side URL transfer library (OpenSSL flavour)
ii  libcurl4-openssl-dev:amd64                      7.68.0-1ubuntu2.2                           amd64        development files and documentation for libcurl (OpenSSL flavour)
ii  libcurlpp-dev:amd64                             0.8.1-2build2                               amd64        c++ wrapper for libcurl (development files)
ii  libcurlpp0:amd64                                0.8.1-2build2                               amd64        c++ wrapper for libcurl
ii  ros-foxy-libcurl-vendor                         2.3.3-1focal.20201006.191618                amd64        Wrapper around libcurl, it provides a fixed CMake module and an ExternalProject build of it.
```
So the full name of the library I am looking for is `libcurlpp-dev`. Next list all the files associated with this lib
```
$ dpkg -L libcurlpp-dev

/.
/usr
/usr/bin
/usr/bin/curlpp-config
/usr/include
/usr/include/curlpp
/usr/include/curlpp/Easy.hpp
/usr/include/curlpp/Easy.inl
/usr/include/curlpp/Exception.hpp
/usr/include/curlpp/Form.hpp
/usr/include/curlpp/Info.hpp
/usr/include/curlpp/Info.inl
/usr/include/curlpp/Infos.hpp
/usr/include/curlpp/Multi.hpp
/usr/include/curlpp/Option.hpp
/usr/include/curlpp/Option.inl
/usr/include/curlpp/OptionBase.hpp
/usr/include/curlpp/Options.hpp
/usr/include/curlpp/Types.hpp
/usr/include/curlpp/cURLpp.hpp
/usr/include/curlpp/internal
/usr/include/curlpp/internal/CurlHandle.hpp
/usr/include/curlpp/internal/CurlHandle.inl
/usr/include/curlpp/internal/OptionContainer.hpp
/usr/include/curlpp/internal/OptionContainer.inl
/usr/include/curlpp/internal/OptionContainerType.hpp
/usr/include/curlpp/internal/OptionList.hpp
/usr/include/curlpp/internal/OptionSetter.hpp
/usr/include/curlpp/internal/OptionSetter.inl
/usr/include/curlpp/internal/SList.hpp
/usr/include/utilspp
/usr/include/utilspp/EmptyType.hpp
/usr/include/utilspp/NonCopyable.hpp
/usr/include/utilspp/NullType.hpp
/usr/include/utilspp/SmartPtr.hpp
/usr/include/utilspp/ThreadingFactoryMutex.hpp
/usr/include/utilspp/ThreadingFactoryMutex.inl
/usr/include/utilspp/ThreadingSingle.hpp
/usr/include/utilspp/ThreadingSingle.inl
/usr/include/utilspp/TypeList.hpp
/usr/include/utilspp/TypeTrait.hpp
/usr/include/utilspp/clone_ptr.hpp
/usr/lib
/usr/lib/x86_64-linux-gnu
/usr/lib/x86_64-linux-gnu/libcurlpp.a
/usr/lib/x86_64-linux-gnu/pkgconfig
/usr/lib/x86_64-linux-gnu/pkgconfig/curlpp.pc
/usr/share
/usr/share/doc
/usr/lib/x86_64-linux-gnu/libcurlpp.so
/usr/share/doc/libcurlpp-dev
```
You can see that there is `/usr/include/curlpp/cURLpp.hpp` listed, which we want to include in the code. For these files to be found, I check the .pc file

```
$ cat /usr/lib/x86_64-linux-gnu/pkgconfig/curlpp.pc

# This is a comment
prefix=/usr
exec_prefix=/usr
includedir=include

Name: curlpp
Description: cURLpp is a libcurl C++ wrapper
Version:                            
Libs: -Llib/x86_64-linux-gnu -lcurlpp -Wl,-Bsymbolic-functions -Wl,-z,relro 
Cflags: -Iinclude 
# libcurl is required as non-private because CurlHandle.inl uses curl_easy_setopt.
Requires: libcurl
```
Based on this I include `curlpp` (`Name:`) in my `CMakeLists.txt`

Not sure if it was a typo, but the command you typed: `sudo apt-get install libcurl-dev`. Shouldn't it be `libcurlpp-dev`?

ipa-hsd, 2020-12-08:
The file associated with `libcurl4-openssl-dev` as seen [here](https://packages.debian.org/stretch/i386/libcurl4-openssl-dev/filelist) is `/usr/include/i386-linux-gnu/curl/curl.h`. I am guessing this is a `C` implementation and not `C++`. So either find a corresponding `C++` implementation for `debian` or change the code to use the `C` api.

wsaihopfsg, 2020-12-08:
curlpp-dev [is indeed not available](https://index.ros.org/d/curlpp-dev/) for Debian 9 (Stretch). 

Thanks for your help.

closed 2020-12-08 by wsaihopfsg