PepperlFuchs/pf_lidar_ros_driver #47: Bug when start_angle & max_num_points_scan are specified

wsaihopfsg, 2021-01-02:
In R2300 there are 2 ways to limit the FOV
  1. User specifies the optional `start_angle` & `max_num_points_scan` arguments when invoking `request_handle_udp`. This is backward-compatible with R2000
  2. By changing the paramaters `measure_start_angle` & `measure_stop_angle`. This is only available in R2300.

The current code makes provision only for case 2 and does not seem to consider case 1, which may result in erroneous RViz output.

For example, consider the following parameters in which the FOV is between -10⁰ to 10⁰
```
scan_frequency = 100Hz
angular_increment = 0.2⁰
measure_start_angle = -50⁰ (default)
measure_stop_angle =  50⁰ (default)
start_angle = -10⁰
max_num_points_scan = 101 
```
RViz output will be wrong
[image]

This is due to the fact that the current code only consider `param_.angle_min`, which is obtained from `measure_start_angle`
https://github.com/PepperlFuchs/ROS_driver/blob/80c423ad038d2ff197bd81588ad40ee3f3cd56e5/pf_driver/src/ros/scan_publisher.cpp#L60

If we use the correct values for `msg->angle_min` & `msg->angle_max`, the RViz output will be okay i.e. scan lines subtended from the origin from -10⁰ to 10⁰
[image]

ipa-hsd, 2022-06-02:
Does https://github.com/PepperlFuchs/pf_lidar_ros_driver/pull/50 fixes this issue?

ipa-hsd, 2022-08-20:
This issue is fixed by https://github.com/PepperlFuchs/pf_lidar_ros_driver/pull/50.

closed 2022-08-20 by ipa-hsd