PepperlFuchs/pf_lidar_ros_driver #62: Maintenance status driver

JaccovdS, 2021-04-19:
I wanted to share with you that we're not that happy with the maintenance status of the driver/repo given our experience over the last year. Here are some examples why:

1. Official release took a very long time, first mention was May 2020 actual date of the first tag (not even ROS release) was November 2020. Reference: https://github.com/PepperlFuchs/pf_lidar_ros_driver/issues/12 https://github.com/PepperlFuchs/pf_lidar_ros_driver/issues/34
2. No Noetic support, also takes quite some time. Reference: https://github.com/PepperlFuchs/pf_lidar_ros_driver/issues/28
3. Meaningfull comments and PR are closed (due to insufficient knowledge?). Reference: https://github.com/PepperlFuchs/pf_lidar_ros_driver/issues/37 https://github.com/PepperlFuchs/pf_lidar_ros_driver/pull/36#issuecomment-736545604
4. Issues stay open for a long time. Reference: https://github.com/PepperlFuchs/pf_lidar_ros_driver/issues/38 

In general what we lack is response within a reasonable timespan and constructive discussions. If you truly want to have an open-source driver we would expect a more active approach in maintaining the driver. 

What is your opinion on this and what are your plans for the future?

@PepperlFuchs @ipa-hsd

JaccovdS, 2021-04-21:
@ipa-hsd I'm in no doubt that a lot of useful work has been done. That's also no point of discussion.

My main point is that you now make your source code publicly available on this platform but you don't seem to be open to open-source development. I.e. accepting / discussing useful PR's, listening to comments / requests / needs, being clear about milestones / timelines / goals. In my opinion open-source development comes with responsibility towards the community. 

We as a company are willing to contribute but based on the past experiences we are faster and better when we take this source and continue on our own. I think this is something you want to prevent as maintainer of an open-source project.

ipa-hsd, 2021-04-21:
That is a very valid point @JaccovdS. We will discuss this with @PepperlFuchs and improve the contributions from the community in the future.

JaccovdS, 2021-04-21:
@ipa-hsd happy to hear. Can you update us on the outcome?

JaccovdS, 2021-06-04:
@ipa-hsd any news?

JaccovdS, 2021-06-28:
@ipa-hsd @PepperlFuchs ping

JaccovdS, 2021-08-17:
@ipa-hsd @PepperlFuchs I think the lack of feedback in this issue proves my point. I will close the issue.

closed 2021-08-17 by JaccovdS