PepperlFuchs/pf_lidar_ros_driver #78: R2000 UHD not supported

robertwil, 2022-05-19:
Hi,
I was testing our R2000 UHD with this driver and it seems it does not work as it says:
[ERROR] [1652961708.518646295]: Device unsupported
[ERROR] [1652961708.521031680]: Unable to initialize device

I debugged a little and found that the function 
PFInterface::handle_version
fails at line 124 (https://github.com/PepperlFuchs/pf_lidar_ros_driver/blob/90213cad3edb4378f379b310a4fd40be6a9dcc47/pf_driver/src/pf/pf_interface.cpp#L124)
which indicated a mismatch of the version as far as I understood.
The actual version inside the function would be 1 (major) 4 (minor)

Is the UHD version not supported and this is a known behavior ?

Firmware: 1.60
Hardware: 1.71
Protocol: 1.04

ipa-hsd, 2022-05-19:
For now could you please change this line manually and recompile? https://github.com/PepperlFuchs/pf_lidar_ros_driver/blob/90213cad3edb4378f379b310a4fd40be6a9dcc47/pf_driver/src/pf/pf_interface.cpp#L108
The way R2000 and R2300 are identified might have to change.

robertwil, 2022-05-20:
Changing the versions leads to a start of the ScanPublisherR2000 without obvious errors .. I could not receive any scan though but this could have different reasons.

ipa-hsd, 2022-05-20:
Can you check with wireshark if you are receiving any packets?

robertwil, 2022-05-23:
I can check the next time I get to the scanner.
At least there are no bigger connectivity issues (like cables, bad network interfaces etc.) as I could get
scans in PACTware as well as in the other non official driver (https://github.com/dillenberger/pepperl_fuchs) but which has other problems.

ipa-hsd, 2022-06-03:
At least to take care of the problem with major and minor version numbers, I have pushed some changes in https://github.com/PepperlFuchs/pf_lidar_ros_driver/pull/77/commits/8234c6f212a8fe622b755bd2e25db8297444a41e, which is part of https://github.com/PepperlFuchs/pf_lidar_ros_driver/pull/77.

ipa-hsd, 2022-07-11:
https://github.com/PepperlFuchs/pf_lidar_ros_driver/pull/77 has been merged and seems to have solved the issue. Please feel free to reopen the issue if you are still facing it.