PepperlFuchs/pf_lidar_ros_driver #79: Device unsupported

guomengnick, 2022-05-23:
Hi
i'm using   
OMD30M-R2000-B23-V1V1D-HD-1L    On Ubuntu20.04 ROS noetic ,
when i launch  " roslaunch   pf_driver   r2000.launch   "
it shows error below:
Device unsupported 
Unable to initialize device 

and i'm sure that i can "ping" Lidar's IP 
thanks

ipa-hsd, 2022-05-23:
This issue sounds similar to https://github.com/PepperlFuchs/pf_lidar_ros_driver/issues/78. What is major and minor versions of your device? 

The current implementation needs to be changed so that it does not rely on the version numbers to identify the device, but checks for the `device_family`. I will make a new PR this week with this update.

guomengnick, 2022-05-24:
Lidar i'm using is this :
[P+F](https://www.google.com/url?q=https://www.pepperl-fuchs.com/taiwan/zh/classid_53.htm?view%3Dproductdetails%26prodid%3D86557&sa=D&source=editors&ust=1653368517523629&usg=AOvVaw32SZqeK0Jl4_90HfZ-6-7_)

and i use "PACTWare " to see Version infomation below:
Firmware: 1.60
Hardware: 1.71
Protocol: 1.04

ipa-hsd, 2022-06-03:
I have pushed some changes in https://github.com/PepperlFuchs/pf_lidar_ros_driver/pull/77/commits/8234c6f212a8fe622b755bd2e25db8297444a41e which is part of https://github.com/PepperlFuchs/pf_lidar_ros_driver/pull/77. Hopefully that should solve the problem.

guomengnick, 2022-06-08:
hi i'm using updated code 
but there is still something wrong
the error code is the same as above 

[ERROR] [1654658163.667808723]: Device unsupported
[ERROR] [1654658163.669342408]: Unable to initialize devicer
[2000_node-3] process has died [pid 125369, exit code 255, cmd /home/ros/pf_ws/devel/lib/pf_driver/ros_main_name:=r2000_node __log:=/home/ros/.ros/log/539ef906-e6d9-11ec-997c-150ca0cd6143/r2000_node-3.log].
log file: /home/ros/.ros/log/539ef906-e6d9-11ec-997c-150ca0cd6143/r2000_node-3*.log

By the way , i use catkin_make to build the code 
thanks.

ipa-hsd, 2022-06-08:
@guomengnick https://github.com/PepperlFuchs/pf_lidar_ros_driver/pull/77 has still not been merged, so if you are using the updated `main` branch, you would still face the error. 

```
and i use "PACTWare " to see Version infomation below:
Firmware: 1.60
Hardware: 1.71
Protocol: 1.04
```
Did you also see `major_version` and `minor_version`?

guomengnick, 2022-06-10:
I don't see any major or minor version you said
my PACTware version  is 5.0 
where is major_version  on the PACTware?

thanks

guomengnick, 2022-06-13:
this is my Lidar configuration :
{
"protocol_name":"pfsdp",
"version_major":1,
"version_minor":4,
"commands":[
"get_protocol_info",
"list_parameters",
"get_parameter",
"set_parameter",
"reboot_device",
"factory_reset",
"reset_parameter",
"request_handle_udp",
"request_handle_tcp",
"feed_watchdog",
"set_scanoutput_config",
"get_scanoutput_config",
"start_scanoutput",
"stop_scanoutput",
"release_handle",
"get_iq_parameter",
"set_iq_parameter",
"list_iq_parameters"
],
"error_code":0,
"error_text":"success"
}

ipa-hsd, 2022-06-14:
In this line
https://github.com/PepperlFuchs/pf_lidar_ros_driver/blob/90213cad3edb4378f379b310a4fd40be6a9dcc47/pf_driver/src/pf/pf_interface.cpp#L108
you need to change it from 3 to 4
```
if (major_version == 1 && minor_version == 4)
```

guomengnick, 2022-06-16:
Okay 
i change to 4
and it works !
Thank you for your help!!

closed 2022-06-16 by ipa-hsd