PepperlFuchs/pf_lidar_ros_driver #91: Possible connection drop after long runtime

abuchegger, 2022-09-27:
Hi!

We had the issue that after running for several hours (overnight), the driver node suddenly stopped publishing data on the scan topic during one of our tests. The node did not output any log messages, and obviously did not try to reconnect to the scanner. When checking, the scanner was reactive (ping + curl). We restarted the node, and it published scan data again.

Have you encountered this issue?
We are using the current main branch with an R2000.

Best regards,
Alex

abuchegger, 2022-10-07:
Hi!

We had the same issue again; please find attached the (filtered) rosout and stdout of the driver node:
[lidar-nav_lidar-rosout.log](https://github.com/PepperlFuchs/pf_lidar_ros_driver/files/9734426/lidar-nav_lidar-rosout.log)
[lidar-nav_lidar-stdout.log](https://github.com/PepperlFuchs/pf_lidar_ros_driver/files/9734427/lidar-nav_lidar-stdout.log)

The last scan data was received at approx. 1665133948, which is about 26 minutes before the appliance (including PC, Lidar, and Ethernet switch) was manually power-cycled. Interestingly, the driver node does notice the connection loss at the time of power off, as can be seen in the rosout.

Best regards,
Alex

ipa-hsd, 2022-10-07:
which version of the code are you using? In the recent times I have not tested the driver by running it for long duration since that issue had been fixed. I will try it out next week. Not sure if https://github.com/PepperlFuchs/pf_lidar_ros_driver/issues/90 is related.

abuchegger, 2022-10-07:
Hi,

we're using the current main branch (commit b07c6170d7112aa5923bb9b1e47c09f865e00e4d). The error occurred approx. 3h48min after the connection to the lidar was established.

Best regards,
Alex

ipa-hsd, 2022-10-07:
Thanks for the info. I am currently running on tight schedule, but will try it with the configuration you mentioned next week.

Hr-Senerity, 2022-10-12:
Sorry to reply so late,

I also use the current main branch (commit b07c617). But during the long run, I had no data outage. However, when the node is shut down and restarted after running for a long time, the output information of the node is #90  and no data output. The R2000 will return to normal until you restart after power failure.

ipa-hsd, 2022-10-13:
@abuchegger and @Hr-Senerity very sorry but looks like I will not be able to take a look at this before leaving for ROSCon next week.

@abuchegger did you monitor the CPU usage when no data was received?

abuchegger, 2022-10-24:
No, unfortunately we have not been monitoring CPU usage. We are planning to integrate that into our software.

We have tried to reproduce the issue with an R2000 in the office, with no luck so far.

Kyros8712, 2024-06-19:
Hi, 
I also encountered the same problem. After using it for a long time, /scan topic will have no data.

Subscribers cannot use callback functions to obtain data.