PepperlFuchs/pf_lidar_ros_driver #92: Release

MichaelGrupp, 2022-10-13:
Hi,

I just wanted to ask what your schedule for a new release looks like. The latest release in the ROS Noetic distro is 1.2.0, and since then there have been quite a number of improvements committed to this repo, which probably would also benefit the user who install the package from the distro. Thanks in advance!

ipa-hsd, 2022-10-13:
@ptruka will be back at office in 1st week of November I think. https://github.com/PepperlFuchs/pf_lidar_ros_driver/pull/86 is the last pending PR to be reviewed before the release.

Arthav24, 2022-11-29:
Hey, Just curious any timeline for release?

RainerKuemmerle, 2023-06-15:
The mentioned PR is merged. Any update / plans on a next release. Thanks.

closed 2023-11-03 by MichaelGrupp