PepperlFuchs/pf_lidar_ros_driver #98: UDP port not used - connection with firewall not possible

robertwil, 2022-11-29:
Hi,

using transport via udp does only work when I disable my firewall (which is unwanted).
Looking at the code it seems, that the udp port set in the config is never really used as it is overwritten by
the port of the local endpoint used in the initial connection to the scanner:
https://github.com/PepperlFuchs/pf_lidar_ros_driver/blob/fc9a38b7348f6979800ee33969c6e7cf1365b34a/pf_driver/src/pf/pf_interface.cpp#L59

So unless I missed something it seems that the port specified by the user is completely ignored which causes the incoming scan data connection to fail if the (random?) used port is not allowed in the firewall.

Can you confirm this behavior and is known/wanted ?

ipa-hsd, 2022-11-29:
@robertwil thanks for pointing this out. This behaviour is not wanted. The code should check if user has provided a port. If not, only then get it from the local endpoint.

ipa-hsd, 2022-11-30:
@robertwil I made some changes locally and tested it with R2000. I will create a PR tomorrow, would be great if you could test it on your side before it is merged (I haven't tried it with a firewall, but just tested if UDP uses the port in the config)

robertwil, 2022-12-01:
@ipa-hsd I can test the PR (unfortunately earliest end of next week). But I guess it should work with the firewall if the specified port is used by the driver and then allowed in the firewall rules.

ipa-hsd, 2022-12-01:
@robertwil https://github.com/PepperlFuchs/pf_lidar_ros_driver/pull/99

robertwil, 2022-12-13:
After a quick test it seems to work as expected (define udp port in config yaml, allow this port in firewall -> there are scans)

closed 2022-12-14 by ipa-hsd