PepperlFuchs/pf_lidar_ros_driver #103: boost error

tbhugra, 2023-02-07:
Starting >>> pf_driver
--- stderr: pf_driver                          
CMake Warning at /usr/share/cmake-3.16/Modules/FindBoost.cmake:1161 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:1283 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:1921 (_Boost_MISSING_DEPENDENCIES)
  /usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:134 (find_package)
  /usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:314 (find_boost)
  /usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:551 (find_external_library)
  /opt/ros/galactic/share/pcl_ros/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/galactic/share/pcl_ros/cmake/pcl_rosConfig.cmake:41 (include)
  CMakeLists.txt:9 (find_package)

CMake Warning at /usr/share/cmake-3.16/Modules/FindBoost.cmake:1161 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:1283 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:1921 (_Boost_MISSING_DEPENDENCIES)
  /usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:134 (find_package)
  /usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:314 (find_boost)
  /usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:551 (find_external_library)
  /opt/ros/galactic/share/pcl_ros/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/galactic/share/pcl_ros/cmake/pcl_rosConfig.cmake:41 (include)
  CMakeLists.txt:9 (find_package)

CMake Warning at /usr/share/cmake-3.16/Modules/FindBoost.cmake:1161 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:1283 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:1921 (_Boost_MISSING_DEPENDENCIES)
  /usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:134 (find_package)
  /usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:314 (find_boost)
  /usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:551 (find_external_library)
  /opt/ros/galactic/share/pcl_ros/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/galactic/share/pcl_ros/cmake/pcl_rosConfig.cmake:41 (include)
  CMakeLists.txt:9 (find_package)

CMake Warning at /usr/share/cmake-3.16/Modules/FindBoost.cmake:1161 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:1283 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:1921 (_Boost_MISSING_DEPENDENCIES)
  /usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:134 (find_package)
  /usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:314 (find_boost)
  /usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:551 (find_external_library)
  /opt/ros/galactic/share/pcl_ros/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/galactic/share/pcl_ros/cmake/pcl_rosConfig.cmake:41 (include)
  CMakeLists.txt:9 (find_package)

CMake Warning at /usr/share/cmake-3.16/Modules/FindBoost.cmake:1161 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:1283 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:1921 (_Boost_MISSING_DEPENDENCIES)
  /usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:134 (find_package)
  /usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:314 (find_boost)
  /usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:551 (find_external_library)
  /opt/ros/galactic/share/pcl_ros/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/galactic/share/pcl_ros/cmake/pcl_rosConfig.cmake:41 (include)
  CMakeLists.txt:9 (find_package)

CMake Error at /usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:62 (message):
  common is required but boost was not found
Call Stack (most recent call first):
  /usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:367 (pcl_report_not_found)
  /usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:551 (find_external_library)
  /opt/ros/galactic/share/pcl_ros/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/galactic/share/pcl_ros/cmake/pcl_rosConfig.cmake:41 (include)
  CMakeLists.txt:9 (find_package)

---
Failed   <<< pf_driver [6.26s, exited with code 1]

Summary: 2 packages finished [13.9s]
  1 package failed: pf_driver
  1 package had stderr output: pf_driver

getting this error when building the driver

ipa-hsd, 2023-02-07:
@tbhugra could you please provide some information about your system (Ubuntu and ROS distro) and what branch / commit hash are you trying to build?

tbhugra, 2023-02-07:
i am using ros2 galactic ubuntu 20.04 and using porting-ros2 branch

ipa-hsd, 2023-02-07:
Can you check the output for 
```
dpkg -s libboost-dev | grep 'Version'
```

tbhugra, 2023-02-07:
.Version: 1.71.0.0ubuntu2

ipa-hsd, 2023-02-07:
OK that's strange. I have the exact same version. I will get back to you.

tbhugra, 2023-02-07:
ok please provide a solution asap i am facing same problem from 3 days

ipa-hsd, 2023-02-07:
Just wanted to let you know that `porting-ros2` branch is still a work in progress. It has not been tested.

tbhugra, 2023-02-07:
okay so any other branch for ros2?

ipa-hsd, 2023-02-07:
No it is an ongoing work. But your support is very much welcome to test and fix bugs. 

I will get get back to you regarding this boost error - it compiles on my PC and also in CI, so I am not sure how to recreate this issue.

tbhugra, 2023-02-07:
ok

ipa-hsd, 2023-05-23:
@tbhugra is this issue fixed for you? I have been unable to reproduce this error.

tbhugra, 2023-05-24:
i have not tried it again, i have used the ros1 code it worked fine for me

tbhugra, 2023-07-27:
hi,
we have built the driver again on ros2 galactic still we are facing the same issue.