PepperlFuchs/pf_lidar_ros_driver #105: Impossible to terminate ros_main node cleanly.

ValerioMa, 2023-02-20:
It is not possible to kill the node cleanly.
Calling `rosnode kill` brings the driver into a deadlock situation and it is not possible to kill cleanly the node.

[image]

ipa-hsd, 2023-02-20:
Yes this is a known issue, but I have ignored it so far since it was not a blocking factor for the current users. Have you been able to debug the problem?

ValerioMa, 2023-02-20:
No, I didn't analyze the problem, but looking at the code this line [here](https://github.com/PepperlFuchs/pf_lidar_ros_driver/blob/main/src/pf_driver/src/ros/ros_main.cpp#L92) seems suspicious.

`net_cv_->wait(net_lock, [&net_fail] { return net_fail; });`

If my understanding is correct here we just wait for a network failure but we do not react to a node termination.

Side note. I see you are using a `ros::AsyncSpinner` and you set the number of threads as the number of processor core (https://docs.ros.org/en/noetic/api/roscpp/html/classros_1_1AsyncSpinner.html#aa5b9b685b7eebebf18ed42697e209897) why?