PepperlFuchs/pf_lidar_ros_driver #107: protocol error: 120 Invalid handle or no handle provided

kalectro, 2023-02-21:
After running the released version (1.2.0) for more than 5 hours, I receive the protocol error
`protocol error: 120 Invalid handle or no handle provided`
This is printed every 60 seconds and no scan data is published. According to the data sheet https://files.pepperl-fuchs.com/webcat/navi/productInfo/doct/doct3469f.pdf?v=20201005101009 this error should be printed when a scan_output command is sent with a bad handle. 
[image]
The way I understand the code, the scan_output command is sent once so I do not know where this could come from.

ipa-hsd, 2023-02-21:
@ptruka any idea about the possible reasons for this?

ipa-hsd, 2023-02-21:
> The way I understand the code, the scan_output command is sent once so I do not know where this could come from.

https://github.com/PepperlFuchs/pf_lidar_ros_driver/blob/aebe0a190b77e3502db4c15d4533ca39de885a8f/src/pf_driver/src/ros/ros_main.cpp#L71-L74

In case of a network failure, the driver tries to reconnect to the device. As far as I remember, there is a limit on the number of times the device can provide new handles. @ptruka correct me if I am wrong.

Has the driver tried to reconnect during this 5 hours run?

kalectro, 2023-02-21:
@ptruka is there a way to read this information from the device through ROS?
@ipa-hsd There is nothing printed by the r2000_node
`[/r2000_node WARNING 1676938854.268373]: Shutdown request received.
[/r2000_node WARNING 1676938854.268851]: Reason given for shutdown: [user request]
[/r2000_node INFO 1676938854.479136]: disconnecting...
[/r2000_node INFO 1676938854.480536]: disconnecting...
[/r2000_node INFO 1676938854.482594]: disconnecting...
[/r2000_node INFO 1676938855.068263]: Device found: OMD30M-R2000-B23-V1V1D-HD-1L
[/r2000_node INFO 1676938855.095618]: Device state changed to Initialized
[/r2000_node INFO 1676938855.149217]: Device state changed to Running
[/r2000_node INFO 1676957815.357771]: protocol error: 120 Invalid handle or no handle provided
[/r2000_node INFO 1676957875.153960]: protocol error: 120 Invalid handle or no handle provided
[/r2000_node INFO 1676957935.154558]: protocol error: 120 Invalid handle or no handle provided
[/r2000_node INFO 1676957995.154212]: protocol error: 120 Invalid handle or no handle provided
[/r2000_node INFO 1676958055.153542]: protocol error: 120 Invalid handle or no handle provided
[/r2000_node INFO 1676958115.155341]: protocol error: 120 Invalid handle or no handle provided
[/r2000_node INFO 1676958175.154177]: protocol error: 120 Invalid handle or no handle provided
[/r2000_node INFO 1676958235.153918]: protocol error: 120 Invalid handle or no handle provided
[/r2000_node INFO 1676958295.153639]: protocol error: 120 Invalid handle or no handle provided
[/r2000_node INFO 1676958355.154243]: protocol error: 120 Invalid handle or no handle provided
[/r2000_node INFO 1676958415.154233]: protocol error: 120 Invalid handle or no handle provided
[/r2000_node INFO 1676958475.153575]: protocol error: 120 Invalid handle or no handle provided
[/r2000_node INFO 1676958535.153705]: protocol error: 120 Invalid handle or no handle provided
[/r2000_node INFO 1676958595.154092]: protocol error: 120 Invalid handle or no handle provided
`

ipa-hsd, 2023-02-21:
```bash
[/r2000_node WARNING 1676938854.268373]: Shutdown request received.
[/r2000_node WARNING 1676938854.268851]: Reason given for shutdown: [user request]
[/r2000_node INFO 1676938854.479136]: disconnecting...
[/r2000_node INFO 1676938855.068263]: Device found: OMD30M-R2000-B23-V1V1D-HD-1L
[/r2000_node INFO 1676938855.095618]: Device state changed to Initialized
```
This looks a bit weird. Why is it reconnecting after the user tried to shutdown? Does this happen often or only during long operation? Can you print out the `info_->handle` here?

https://github.com/PepperlFuchs/pf_lidar_ros_driver/blob/aebe0a190b77e3502db4c15d4533ca39de885a8f/src/pf_driver/src/pf/pf_interface.cpp#L94

kalectro, 2023-02-21:
@ipa-hsd we use the respawn=True argument in our launch files, this is expected behavior. I manually killed the node to test https://github.com/PepperlFuchs/pf_lidar_ros_driver/pull/106

@ptruka 
"product":"OMD30M-R2000-B23-V1V1D-HD-1L",
"part":"305986",
"revision_fw":"1.62",
"revision_hw":"1.72",

ipa-hsd, 2023-02-21:
So the time difference between 
```
[/r2000_node INFO 1676938855.149217]: Device state changed to Running
```
and 
```
[/r2000_node INFO 1676957815.357771]: protocol error: 120 Invalid handle or no handle provided
```
is 5 hours?

kalectro, 2023-02-21:
yes

MichaelGrupp, 2023-02-23:
Thanks for the fast fix! Is this going to be included in a (patch) release in the near future?