PepperlFuchs/pf_lidar_ros_driver #110: transplant to ros2

dlutjwh, 2023-03-06:
hi, thanks for your work, when i  transplant the code to ros2, which can connet to the r2000 like this:
[r2000_node-4] [INFO] [1677779762.428661354] [r2000_node]: Device found: R2000
[r2000_node-4] [INFO] [1677779762.445970648] [r2000_node]: Device state changed to Initialized
[r2000_node-4] [INFO] [1677779762.452387113] [r2000_node]: Device state changed to Running
but after that return fault:
[r2000_node-4] terminate called after throwing an instance of 'rcpputils::IllegalStateException'
[r2000_node-4]   what():  Wrongly initialized. Serialized message has a capacity of zero.
dou you have any idea for that，thanks a lot.

ipa-hsd, 2023-03-06:
ROS 2 version is still not ready to use. I am currently trying to fix the de-serialization issue - they way messages are serialized in ROS 2 is different than in ROS 1. Any help / suggestions are welcome

closed 2023-03-07 by dlutjwh

emoPointer, 2024-07-13:
Has the problem been solved？I have the same problem，
log is below:
```
[ros_main-1] protocol error: 110 Unknown parameter 'skip_scans'
[ros_main-1] [INFO] [1720847655.008640679] [ros_main]: Device state changed to Initialized
[ros_main-1] [INFO] [1720847655.019496512] [ros_main]: Device state changed to Running
[ros_main-1] [ERROR] [1720847657.009863096] [transport]: Time out: No packets received in 2 seconds
[ros_main-1] [ERROR] [1720847657.010546774] [ros_main]: Network failure
[ros_main-1] [INFO] [1720847657.047467382] [ros_main]: Device state changed to Uninitialized
[ros_main-1] [INFO] [1720847657.094887568] [ros_main]: Device found: R2000
[ros_main-1] protocol error: 110 Unknown parameter 'skip_scans'
[ros_main-1] [INFO] [1720847657.141741912] [ros_main]: Device state changed to Initialized
[ros_main-1] [INFO] [1720847657.153894682] [ros_main]: Device state changed to Running
[ros_main-1] [ERROR] [1720847659.148992255] [transport]: Time out: No packets received in 2 seconds
[ros_main-1] [ERROR] [1720847659.149705975] [ros_main]: Network failur
```