PepperlFuchs/pf_lidar_ros_driver #116: Unable to communicate with device

thkfighter, 2023-04-18:
I am using 

- ROS 2 foxy
- pf_lidar_ros_driver, branch=porting-ros2
- Pepperl Fuchs LiDAR OMD60M-R2000-B23-V1V1D-1L

An error occurs, saying "Unable to communicate with device. Please check the IP address". 
Computer IP: 10.0.10.100
LiDAR IP: 10.0.10.9
Pinging the LiDAR from the computer works.

Console log:
```
$ ros2 launch pf_driver r2000.launch.py 
[INFO] [launch]: All log files can be found below /home/tao4sgh/.ros/log/2023-04-18-15-26-47-884535-SZHM-C-0001Y-160661
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros_main-1]: process started with pid [160671]
[INFO] [robot_state_publisher-2]: process started with pid [160673]
[INFO] [rviz2-3]: process started with pid [160675]
[robot_state_publisher-2] Parsing robot urdf xml string.
[robot_state_publisher-2] Link scanner_link had 0 children
[robot_state_publisher-2] [INFO] [1681802808.062767329] [robot_state_publisher]: got segment scanner_link
[robot_state_publisher-2] [INFO] [1681802808.062843581] [robot_state_publisher]: got segment world
[ros_main-1] [INFO] [1681802808.070880389] [ros_main]: device name: R2000
[ros_main-1] [INFO] [1681802808.070973082] [ros_main]: transport_str: udp
[ros_main-1] [INFO] [1681802808.070988987] [ros_main]: scanner_ip: 10.0.10.9
[ros_main-1] [INFO] [1681802808.071013201] [ros_main]: port: 
[ros_main-1] [INFO] [1681802808.071029250] [ros_main]: scan_frequency: 2
[ros_main-1] [INFO] [1681802808.071044094] [ros_main]: samples_per_scan: 0
[ros_main-1] [INFO] [1681802808.071061262] [ros_main]: start_angle: -1800000
[ros_main-1] [INFO] [1681802808.071076083] [ros_main]: max_num_points_scan: 0
[ros_main-1] [INFO] [1681802808.071090434] [ros_main]: watchdogtimeout: 60000
[ros_main-1] [INFO] [1681802808.071104318] [ros_main]: watchdog: 1
[ros_main-1] [INFO] [1681802808.071118090] [ros_main]: num_layers: 1
[ros_main-1] [INFO] [1681802808.071134108] [ros_main]: topic: /scan
[ros_main-1] [INFO] [1681802808.071148312] [ros_main]: frame_id: scanner_link
[ros_main-1] [INFO] [1681802808.071162061] [ros_main]: packet_type: C
[ros_main-1] [INFO] [1681802808.071176306] [ros_main]: apply_correction: 0
[ros_main-1] [INFO] [1681802808.071190152] [ros_main]: start_angle: -1800000
[rviz2-3] [INFO] [1681802808.974008982] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1681802808.974150852] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-3] [INFO] [1681802809.061155511] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [ERROR] [1681802809.230101812] [rviz2]: PluginlibFactory: The plugin for class 'rviz_common/Time' failed to load. Error: According to the loaded plugin descriptions the class rviz_common/Time with base class type rviz_common::Panel does not exist. Declared types are  nav2_rviz_plugins/Navigation 2 slam_toolbox::SlamToolboxPlugin
[rviz2-3] Parsing robot urdf xml string.
[ros_main-1] HTTP ERROR: Empty reply from server
[ros_main-1] [ERROR] [1681802848.105734579] [ros_main]: Unable to communicate with device. Please check the IP address
[ros_main-1] [ERROR] [1681802848.105953838] [ros_main]: Unable to initialize device
[ros_main-1] terminate called without an active exception
[ERROR] [ros_main-1]: process has died [pid 160671, exit code -6, cmd '/home/tao4sgh/ws_pf_ros2/install/pf_driver/lib/pf_driver/ros_main --ros-args -r __node:=ros_main --params-file /home/tao4sgh/ws_pf_ros2/install/pf_driver/share/pf_driver/config/r2000_params.yaml'].
```

ipa-hsd, 2023-04-18:
I don't think this is an IP address problem.
```
HTTP ERROR: Empty reply from server
```
The HTTP request failed. @ptruka ?

chenwu-cc, 2023-05-22:
> 我将在下周使用相同的设备对其进行测试并报告。

May I ask if your driver implementation has been successful? Can you share it

ipa-hsd, 2023-10-17:
@thkfighter closing this issue. If you are still facing the issue, feel free to re-open

closed 2023-10-17 by ipa-hsd