PepperlFuchs/pf_lidar_ros_driver #119: Restarting the node several times starts causing persistent protocol error

sarguez, 2023-10-25:
Hello,

When I kill and restart the node several times, I start getting this after the third one:

```
[/r2000_node  INFO 1698241970.008605855]: Device found: R2000
protocol error: 120 New connection would exceed limit on number of connections(3>=3)
[/r2000_node ERROR 1698241970.015934572]: Could not acquire communication handle
[/r2000_node ERROR 1698241970.015972343]: Unable to initialize device
```

This problem persists until I powercycle the scanner itself.

I suspect that there may be some connection left to the laser that is not cleaned up upon killing the node.

ipa-hsd, 2023-10-25:
Thanks for reporting the issue. Which ROS distro and which branch did you test the driver with?

sarguez, 2023-10-25:
Hello, I tested with ROS noetic, on tag: 1.2.1

ipa-hsd, 2023-10-27:
@sarguez fixed in https://github.com/PepperlFuchs/pf_lidar_ros_driver/pull/120. Could you please confirm?

sarguez, 2023-10-27:
@ipa-hsd  Hello, thanks for the quick fix. I tested it now as well, it works.

ipa-hsd, 2023-10-27:
Actually the "fix" has caused an issue with the reconnect feature. I am working on it. Once that feature is restored, I will close the issue.

ipa-hsd, 2023-10-30:
The reconnect feature is stored.

closed 2023-10-30 by ipa-hsd